MaslowCNC/Maslow_4 #59: Feature Request: Firmware update from github.

JDogHerman, 2024-03-09:
On the Maslow4 it would be great to have an option where the FluidNC could check the firmware version vs github and if a newer release is available provide and option to update the firmware along with the index.html file. This feature wouldn’t really work for Maslow4 setups that arnt connected to the internet but those that are could be kept up to date easily without an intermediate download of the files from github.

Maybe the github repo could be a variable in the maslow.yaml allow individuals to point to the repo they might want to check against. This would deal with branches and forks.

BarbourSmith, 2024-03-09:
I think that this is a fantastic idea

davidelang, 2025-09-19:
created issue #331 to see if copilot can write the code for this.

davidelang, 2025-10-01:
fixed by #332