MaslowCNC/Maslow_4 #67: Feature Request: Helpful messages/translations

JDogHerman, 2024-03-14:
In addition to the serial message like "error: 8" for the machine is in alarm mode and needs to be disarmed first, maybe there could be a translation saying that the alarm is engaged and telling the user to disengage the alarm before the machine will move?

[image]

BarbourSmith, 2024-03-20:
100% 

A lot of this is coming straight from FluidNC so we should re-evaluate what it means for us.

To be honest I don't think that we should even be starting out in the Alarm state. That makes sense on big machines where it's a safety issue, but for us I think it creates more confusion than help.

davidelang, 2025-09-21:
given that really bad things will happen if you try to move wihtout knowing where the belts are, I think that it's reasonable to start in an alarm state.

with the current startup, being in alarm until we have retracted, extended, and applied tension is a very reasonable thing.
Even once we save the belt lengths, we can't know if we had a clean shutdown or a panic (if we write to disk/flash/nvm ever time we enter/exit the idle state we could do it, but would we wear out the storage media?

BarbourSmith, 2025-09-22:
We have our own state machine that won't let you say jog in the XY direction until after the belts are extended and tension is applied. I think that the only thing that the alarm state is doing right now is locking the z-axis from moving.

What if we replaced where it says "Alarm" with our own state (curently shown at the top of the settings panel:

[image]

I think that information is more useful to us.