MaslowCNC/Maslow_4 #95: Feature: UI: Z position enhancement

ronlawrence3, 2024-04-10:
I see you don't have issues turned on in the UI project, so I'll put this here. As one of the forum posts I was in mentioned, on the Z dialog it would be nice to have a "z-top" button that would send Z all the way to the top so the user can change bits.  I don't know if we would save the max Z during calibration or as another step during calibration, but it would be nice to have this.

davidelang, 2025-09-21:
I like the idea, I don't know if it can be done just in the UI (how does it know how high it can go? we could have that be a maslow.yaml or preferences.yaml parameter??)

davidelang, 2025-09-21:
now that we can file issues in the UI project, I created https://github.com/BarbourSmith/ESP3D-WEBUI/issues/201 for this request.

davidelang, 2025-10-07:
@ronlawrence3 can you check if the PR linked to above does what you are looking for?

closed 2025-12-23 by github-actions[bot]