MaslowCNC/Maslow_4 #147: Canvas for gcode not drawing correct bounds

ronlawrence3, 2024-06-30:
From the code, the 4 camera angles are: (click the image to cycle between them)
1. oblique
2. oblique with bounds
3. top
4. top with bounds
5. top with bounds and belts? 

All but 1 and 3 seem like they are not really correct in terms of the view area and of the machine bounds.  

See

BarbourSmith, 2024-07-01:
100% an issue. The bounds have an issue with loading the machine dimensions I think because I tried to write the UI so that the folks at FluidNC could use it if they wanted it merged back in, but it seems like they don't so at this point we could probably hard code those numbers.

I would also love to add a view which is a more zoomed in view which follows the machine so the purple dot is always in the center of the screen and it pans as the machine moves.

BarbourSmith, 2025-09-22:
No, I think this issue is still happening. I worked on fixing it recently, but I don't think that I was successful.

The issue is that right now the machine is showing the bounding box of the 4x8 sheet as centered on the home position, when it should really be centered between the anchor points and shouldn't move just because the home position is redefined.