MaslowCNC/Maslow_4 #152: Feature: Progress bar shown when cutting

BarbourSmith, 2024-08-15:
### Machine Context

It would be great if we could show the percentage complete while running a cut on the main tab somewhere

### Feature Description

Not sure where it would fit on the UI, but something to think about

### Other Approaches

It already exists on the other tab so it shouldn't be too hard to mvoe over 

### How I Can Help

I will make it :p

spnc-omz, 2024-08-16:
I'd love to help with UI enhancements but the documentation on how it's all built isn't really there. how can I get involved?

ronlawrence3, 2024-08-16:
The repo for the ui is [here](https://github.com/BarbourSmith/ESP3D-WEBUI). It is an interesting environment to code in... reach out on the forums if you have trouble getting started and we'll all chime in. there are a few topics in there about dev environments, etc.