MaslowCNC/Maslow_4 #156: Problem: "Start Probe" Button Produces Incorrect GCode

andith00, 2024-10-09:
### Controller Board

M4

### Machine Description

M4

### Configuration file

```yaml
N/A
```

### Startup Messages

```text
N/A
```

### User Interface Software

WebUI

### What happened?

After enabling the probe options under Preferences on maslow.local, the probe options appear under the FluidNC Controller tab.  When you click "Start Probe", a GCode command is automatically executed in the Command prompt below.  The GCode that it executes is "G38.6" (plus parameters).  This should be "G38.2" (plus parameters).  As a result, the executed code just creates an error.  Manually entering the "G38.2" (plus the same parameters) into the command prompt, however, causes the machine to begin probing correctly (I can confirm that a probe triggered by G38.2 works exactly as intended).

BarbourSmith, 2024-10-23:
I went to fix this and found the relevant lines of code, but there is a comment which gives me pause:
```

function StartProbeProcess() {
  // G38.6 is FluidNC-specific.  It is like G38.2 except that the units
  // are always G21 units, i.e. mm in the usual case, and distance is
  // always incremental.  This avoids problems with probing when in G20
  // inches mode and undoing a preexisting G91 incremental mode
  var cmd = 'G38.6 Z-'
```

It's a super easy change to make switching that to .2, but will it break things in inches mode?

andith00, 2024-10-23:
Interesting.  When clicking “Start Probe”, the command “G38.6 Z-50 F60 P0” is automatically entered into the command window.  Z, F, and P are just parameters depending on your settings.  Immediately, it gives me the error “Error 20” and the next line “Unsupported or invalid g-code command found in block.” And on the next line, it says “Error  :  undefined  :undefined”.

andith00, 2024-10-23:
Could it be the “Z-“ in the var cmd that is messing things up?

BarbourSmith, 2024-10-24:
There are some changes in the latest version of FluidNC that we haven't incorporated it. Let's try changing it to the standard G38.2 and we can revisit if we run into any issues

BarbourSmith, 2024-10-24:
Here is an updated version, will you give this a test and let me know if it fixes the issue?

[index.html.gz](https://github.com/user-attachments/files/17513129/index.html.gz)

andith00, 2024-10-25:
Tested new file and worked flawlessly.

BarbourSmith, 2024-10-25:
Does it matter if the machine is in inches or mm mode when you run it?

andith00, 2024-10-28:
Tested; it does not matter.  Works perfectly in either machine setting.  The probe settings are in metric regardless of machine settings.

BarbourSmith, 2024-10-28:
Awesome, I'll merge it in and it should work in the next release. Thanks for the help!

andith00, 2024-10-29:
The Start Probe button actually incorporates significantly more than using a G38.2 on its own.  The button adds in additionally functionality.  With the button, the machine will set the new Z-home for you automatically, and it will retract the Z back up after making contact to a safety position.  I checked the accuracy with which it automatically set the Z-home by lowering down in .1mm increments until contact was made again, and got the exact same Z position as it had automatically set after clicking Start Probe.

It's great.  The Z sets itself with a single button click.

BarbourSmith, 2024-10-29:
Wonderful! Very exciting stuff 😀