MaslowCNC/Maslow_4 #180: Problem: when starting gcode while not calibrated, WebUI disconnects

JustusRijke, 2025-01-01:
### Controller Board

Maslow 4.0

### Machine Description

Maslow 4.0

### Configuration file

```yaml
board: Maslow
name: Maslow S3 Board
meta: 
Maslow_vertical: false
Maslow_calibration_grid_width_mm_X: 1500.000000
Maslow_calibration_grid_height_mm_Y: 800.000000
Maslow_calibration_grid_size: 7
Maslow_tlX: -3.500000
Maslow_tlY: 3516.500000
Maslow_trX: 4190.600098
Maslow_trY: 3510.300049
Maslow_blX: 0.000000
Maslow_blY: 0.000000
Maslow_brX: 4194.500000
Maslow_brY: 0.000000
Maslow_tlZ: 100.000000
Maslow_trZ: 56.000000
Maslow_blZ: 34.000000
Maslow_brZ: 78.000000
Maslow_Retract_Current_Threshold: 1300
Maslow_Calibration_Current_Threshold: 1200
Maslow_Acceptable_Calibration_Threshold: 0.450000
Maslow_beltEndExtension: 30.000000
Maslow_armLength: 123.400002
stepping: 
  engine: Timed
  idle_ms: 240
  pulse_us: 4
  dir_delay_us: 0
  disable_delay_us: 0
  segments: 12

uart1: 
  txd_pin: gpio.1
  rxd_pin: gpio.2
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 115200
  mode: 8N1

spi: 
  miso_pin: gpio.13
  mosi_pin: gpio.11
  sck_pin: gpio.12

sdcard: 
  cs_pin: gpio.10
  card_detect_pin: NO_PIN
  frequency_hz: 8000000

kinematics: 
  Cartesian: 

axes: 
  shared_stepper_disable_pin: NO_PIN
  shared_stepper_reset_pin: NO_PIN
  x: 
    steps_per_mm: 80.000000
    max_rate_mm_per_min: 2000.000000
[... 302 lines cut ...]
Grbl 0 [FluidNC v0.87 (Maslow-Main-ec171155-dirty) (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
[MSG:INFO: Encoder connected on Top Left]
[MSG:INFO: Motor detected on Top Left]
[MSG:INFO: Encoder connected on Top Right]
[MSG:INFO: Motor detected on Top Right]
[MSG:INFO: Encoder connected on Bottom Left]
[MSG:INFO: Motor detected on Bottom Left]
[MSG:INFO: Encoder connected on Bottom Right]
[MSG:INFO: Motor detected on Bottom Right]
[MSG:INFO: Current z-axis position loaded as: 0.000]
[MSG:INFO: Starting Maslow Version 0.87]
[MSG:INFO: Channel auto report interval set to 50 ms]
MIN"tr": -0.011,"br": 0.000,"bl": 0.000,"etl": 0.000,"etr": -0.011,"ebr": 0.000,"ebl": 0.000,"extended": false}
```

### What happened?

Steps to reproduce:
- turn on Maslow
- connect to WebUI
- upload Maslow logo gcode file
- click "Play" 

Fan starts to run, shortly after WebUI disconnects and cannot be reached. Device still connected to network. Also still responsive when using fluidterm. Powercycle needed to reset.

BarbourSmith, 2025-01-06:
Excellent bug catch!

davidelang, 2025-09-21:
has this been fixed? There has been so much change since this was submitted...