MaslowCNC/Maslow_4 #181: Improvement: update AS5600 library (magnetic encoder)

JustusRijke, 2025-01-03:
### Machine Context

The AS5600 library has been [updated](https://github.com/RobTillaart/AS5600/blob/master/CHANGELOG.md) since the version currently used (0.34). Updates include error handling that could help improve I2C error handling for the Maslow.

### TODO
1. Investigate possible breaking changes mentioned (see changelog, v0.50)
2. Update library and test with (multiple) Maslows
3. Integrate AS5600 library error handling (and other features if they add value)

### Feature Description

-

### Other Approaches

-

### How I Can Help

-

BarbourSmith, 2025-01-03:
Excellent suggestion! 

I believe that I modified the AS5600 library a bit over time, but it shouldn't be too hard to bring in the new code

JustusRijke, 2025-01-07:
This is the diff between the original v0.3.6 and the Maslow version:

```diff
+#include "../Report.h"
 #include "AS5600.h"
 
 //  CONFIGURATION REGISTERS
@@ -375,19 +376,20 @@ float AS5600::getAngularSpeed(uint8_t mode) {
 int32_t AS5600::getCumulativePosition() {
   int16_t value = readReg2(AS5600_RAW_ANGLE) & 0x0FFF;
 
-  //  whole rotation CW?
-  //  less than half a circle
-  if ((_lastPosition > 2048) && (value < (_lastPosition - 2048))) {
-    _position = _position + 4096 - _lastPosition + value;
+  // A transition from the highest 1/4th to the loweset 1/4th
+  if ((value < 1024) && (_lastPosition > 3072)) {
+    _position += abs(4096 - abs(value - _lastPosition));
+  }
+  // A transition from the lowest 1/4th to the highest 1/4th
+  else if ((_lastPosition < 1024) && (value > 3072)) {
+    _position -= abs(4096 - abs(value - _lastPosition));
+  }
+  // We've just moved a bit up
+  else {
+    _position += (value - _lastPosition);
   }
-  //  whole rotation CCW?
-  //  less than half a circle
-  else if ((value > 2048) && (_lastPosition < (value - 2048))) {
-    _position = _position - 4096 - _lastPosition + value;
-  } else
-    _position = _position - _lastPosition + value;
-  _lastPosition = value;
 
+  _lastPosition = value;
   return _position;
 }
 ```

Bit easier to see this way:
[image]

The `#include "../Report.h"` doesn't appear to be needed. The position calculation however is pretty core functionality, and is still the same in the current version of the AS5600. @BarbourSmith do you remember why you changed this code?

BarbourSmith, 2025-01-07:
I have a vague memory of it being a bug, but I can't recall what it was and unfortunately git blame isn't telling us much.

I vaguely remember that there was some special case where the original code could lose one full rotation accidentally 

It looks like I switched it so that the transition has more threshold for when a full rotation is counted

JustusRijke, 2025-01-07:
Ok, we definitely have to do a proper test with actual hardware for this one then; it'll have to wait until I have a stable setup.